thaytan/OpenHMD #8: Cross-platform support

thaytan, 2020-02-10:
In the interests of iterating fast, I've probably used posix or GNU specific API in places. Getting things running on Windows and other unixes needs testing.

UVC camera access is done directly through libusb, which should work cross-platform but again - needs testing.

TheOnlyJoey, 2020-02-10:
Currently libusb is not really meant to be used directly (driver only use it through hidapi-libusb), and having a hard dependency against libusb would mean losing support for hidraw only.

A solution would be looking into a first draft implementation of libconvergence to replace hidapi/libuvc/libusb. Another would be platform testing to see how big of a issue this would be for short term.

thaytan, 2020-02-11:
The USB interfaces we need to talk to are not HID interfaces, so hidapi is not an option. I'm not sure that it excludes hidapi-hidraw mode - that will still be used for talking to the HMD, and libusb will be used to talk directly to the (completely separate) camera sensor.

mironovil, 2020-03-20:
Hello. I am linux user. I have oculus rift CV1 with oculus touch. Can I help you with something?

thaytan, 2020-03-20:
Hi! This issue is about making sure that the new code (particularly the USB camera sensor access) is portable to Windows and MacOS - less so with Linux, which is where I'm doing my development.

Take a look at the other issues I've filed and see if anything looks approachable for you. Otherwise, just testing things and reporting your experience is useful too. Don't expect too much yet though - it's early development still!

I'm currently working on the https://github.com/thaytan/OpenHMD/tree/rift-correspondence-search but I think I'll do some consolidation of all the work so far to a new clean development branch soon.

thaytan, 2021-04-20:
I pushed a couple of fixes for MSVC (like the byteorder.h include), but it's not enough to get things building. The UVC driver directly uses the posix monotonic clock - an API that Windows doesn't have. It will need to go through the OpenHMD compat call, but that made me realise that there's probably more wrong with the timing handling on Windows around that.

sid-cypher, 2021-04-20:
Apologies if I'm ignorantly bringing up wrong/irrelevant stuff, but a cursory search gives examples like [profileapi.h QueryPerformanceCounter](https://docs.microsoft.com/en-us/windows/win32/api/profileapi/nf-profileapi-queryperformancecounter) for the monotonic clock thing, and there's stuff like cross-platform libuvc, too.

Don't know when I'll get to figuring out OpenHMD code, but I do love the idea of not only having free software drivers, but knowing them enough to play around with changes, too.

thaytan, 2021-04-20:
OpenHMD has wrapper function for getting monotonic clock time already, I just didn't use it in that spot because it was awkward.

This simple UVC driver is basically doing what libuvc does, but allows me to maintain a queue of zero-copy framebuffers, and to talk directly to the USB device to send the radio commands that makes the cameras listen for exposure triggers from the headset.