thaytan/OpenHMD #11: Pose matching: Add an LED size model

thaytan, 2020-03-10:
If we use information about the expected size of an LED at a given distance, we can more intelligently constrain the match distance for blobs to LEDs in rift-sensor-pose-helper.c

Inotamira, 2020-10-26:
Would these equations assist with that?

https://pages.uoregon.edu/soper/Light/luminosity.html

thaytan, 2020-10-27:
What I had in mind when I wrote this is a model that would take into account the luminosity of the LED, the physical size, the distance and angle to the camera in the proposed pose, and spit out an expected brightness range for the pixels, and the size of the bounding box in pixels, and possible also a 'shape' to match against. When checking if the camera-observed pixels match a proposed pose, it would be good to know that we expect an LED to only appears as 1-2 pixels for example, and not match against a blob that's many times bigger than that.

That's definitely in 'future refinement' territory though - it'll produce more accurate and stable pose observations, but isn't critical to "getting something working" at this stage.

Doomsdayrs, 2021-02-17:
Would this have to be done via mathhh, or would we be able to help by gathering data?

thaytan, 2021-02-17:
Maths mostly, and definitely an item for the distant future still

thaytan, 2021-03-29:
9f118cd9970d7cf00324a4a25e25ae15ccefabd2 adds some prediction of the expected size of LEDs at a given distance. Hard to tell if it helps yet, but it didn't seem to hurt.