thaytan/OpenHMD #13: Pose matching: Better ways to disambiguate controllers

thaytan, 2020-03-10:
The current correspondence matching gets easily confused as to which controller is which, because the LED arrangement is quite similar around the ring.

Inotamira, 2020-10-26:
It's been a while since I looked into how it specifically works, but, from what I recall, the way that Oculus does it, is by flash codes to "sync" things up, but we don't really have those and you would have to work them out yourself. If that's not the case, you could probably get this working by having it examine the angle that the most extreme left or right respectively top 12 LEDs are coming in at. Since the rings are arched in opposing manners, you should be able to isolate them based on what side is facing the tracking camera(s) and the relative position of the HMD itself.

thaytan, 2020-10-27:
There aren't any blink codes on the touch controllers, and actually Oculus don't seem to use them on the headset since about the time the touch controllers were released.

I haven't updated this tracker for a while, so it's a bit out of date. The code has gotten better at disambiguation since I filed it through a combination of improvements. A big one is using the gravity vector from the IMU as a filter on plausible poses for each controller.

closed 2021-10-24 by thaytan