thaytan/OpenHMD #15: Would you be able to do a development sprint to get very rudimentary CV1-SteamVR working?

hanaral, 2020-06-20:
This project has been steadily improving since I first saw it last December!

I was wondering if you would be able to get the very basics of the CV1 to work with steamVR, as that seems to have garnered the most attention from the large amount of people with CV1's wanting to switch fully to Linux.

Being able to promote the progress you've made will definitely gather even more attention. The youtube channel LinusTechTips (11.2 million subscribers) talked about VR in the new Linux gaming video they published a couple of days ago, and the basic OpenHMD drivers that Rift's can use. I'd happily promote this on the Linux gaming subreddit to gather more funding for the project! (And maybe the project will get mentioned in the next Linux video LTT publishes!)

That being said, due to the current situation it is perfectly reasonable to not do this as I do not know about how much more time you can spare.

thaytan, 2020-06-20:
Definitely possible. I made some progress on that in March in the https://github.com/ChristophHaag/SteamVR-OpenHMD/tree/controller-input branch. It's incomplete, but controllers are starting to work a little in that branch.

It might be possible to do a binary release that can be easily installed by end users on most distros

hanaral, 2020-06-20:
Great! I asume the controllers requires more effort as they aren't directly communicating with the drivers, but at least there is progress. 

Having pre built packages would be very helpful, especially for those who haven't got experience with compiling or the terminal. I suggest starting with a .deb and a tarball for people who want to package it themselves.

thaytan, 2020-10-05:
My latest video update at https://www.youtube.com/watch?v=cQaOpG0Oek0 has some SteamVR action. It's not usable yet, but it's getting there.

thaytan, 2021-03-16:
I think I'll consider this closed now. It's still very much under development, but I think we're past the "Rudimentary support" stage now. 

https://twitter.com/thaytan/status/1371736287594442756

closed 2021-03-16 by thaytan