thaytan/OpenHMD #16: position-hack branch fails to build with ninja

hanaral, 2020-06-20:
Ninja fails to build with the error being:
`[1/5] Compiling C++ object 'openhmd@sha/src_drv_oculus_rift_rift-sensor-opencv.cpp.o'.
FAILED: openhmd@sha/src_drv_oculus_rift_rift-sensor-opencv.cpp.o 
c++ -Iopenhmd@sha -I. -I.. -I.././include -I/usr/include/hidapi -I/usr/include/libusb-1.0 -I/usr/include/opencv -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -g -fPIC -pthread -fvisibility=hidden -DDRIVER_OCULUS_RIFT -DHAVE_OPENCV=1  -MD -MQ 'openhmd@sha/src_drv_oculus_rift_rift-sensor-opencv.cpp.o' -MF 'openhmd@sha/src_drv_oculus_rift_rift-sensor-opencv.cpp.o.d' -o 'openhmd@sha/src_drv_oculus_rift_rift-sensor-opencv.cpp.o' -c ../src/drv_oculus_rift/rift-sensor-opencv.cpp
../src/drv_oculus_rift/rift-sensor-opencv.cpp: In function ‘void refine_pose(double**, rift_led**, int, quatf*, vec3f*, double*)’:
../src/drv_oculus_rift/rift-sensor-opencv.cpp:260:7: error: ‘solvePnPRefineLM’ is not a member of ‘cv’
   cv::solvePnPRefineLM (list_points3d, list_points2d, dummyK, dummyD, rvec, tvec);
       ^~~~~~~~~~~~~~~~
../src/drv_oculus_rift/rift-sensor-opencv.cpp:260:7: note: suggested alternative: ‘solvePnPRansac’
   cv::solvePnPRefineLM (list_points3d, list_points2d, dummyK, dummyD, rvec, tvec);
       ^~~~~~~~~~~~~~~~
       solvePnPRansac
ninja: build stopped: subcommand failed.
`

thaytan, 2020-06-20:
That means that your OpenCV is too old (and that the meson configuration version check is incorrect).

I think solvePnPRefineLM was added in OpenCV 3.4

thaytan, 2020-10-05:
This should be working now - I added a check to compile against older OpenCV some time ago.

closed 2020-10-05 by thaytan