thaytan/OpenHMD #17: UVC Dropping short frame warnings

LameMonster82, 2020-07-25:
I've been trying to get the dev-oculus-position-hack branch to work with my CV1 headset however the following is getting constantly spammed in the log:
```
UVC Dropping short frame: 1220184 < 1228800 (8616 lost)
UVC PTS changed in-frame at 1208996 bytes. Lost 73 ms
UVC Dropping short frame: 1208996 < 1228800 (19804 lost)
UVC PTS changed in-frame at 1211908 bytes. Lost 51 ms
```
The same will happen in the OpenHMD-RiftPlayground too.
I'm not quite sure why however in the opengl example my headset will also start drifting away  into a random direction and it may be because of that.

It may be because im using an unsupported USB 3 card from VIA Labs and if I plug my sensors into a USB 2 port i will get the error:
```
failed to submit iso transfer 0. Error -2
could not start streaming
```
Thank you in advance!

thaytan, 2020-07-26:
The short frame warnings happen when the image analysis takes too long - because it all happens in a single thread at the moment. I plan at some point to split the capture and analysis into several threads. Frames will still occasionally get dropped then, if things are too slow, but it'll be clean entire frames and not random USB packets.

The 2nd error happens because when the sensors are plugged into a USB2 port, they only offer JPEG compressed video frames, and I haven't written any code to support capture+decode of JPEG yet.

The "drifting away" happens when the tracking finds the headset in the cameras for a short moment and then loses it again - without the feedback from the cameras, the IMU interpolation loses position really rapidly. The filtering that I'm working on at the moment will help with that a little, by extracting better bias values and by tracking the variance in the position to know when the position is so uncertain that it should not be reported any more.

LameMonster82, 2020-07-28:
Thank you for the explanation. I read your blog post about OpenHMD and the Oculus rift which gave me a basic idea of how the IMU implementation works and what is used for. However since it only provides orientation data isn't it better to "lock" the headset/controllers in one place (until detected by the sensors) rather than allowing them to "drift away"? It may help with ghosting in the future and prevent further motion sickness.

Also another question. Is it possible to get a view of the rift sensor and what the driver recognizes as the LEDs? Like the OpenHMD-RiftPlayground demo but with updated code. The one in the demo seems to be very old and funky with challenging environments.

closed 2020-07-28 by LameMonster82

thaytan, 2020-07-29:
What's happening is that initially the IMU *is* only used for 3DOF orientation - until the computer vision that's looking for the devices in the sensor data detects a solid lock. At that point, it starts trying to interpolate position using the IMU to help with the tracking. However, because it's all still under development, there's logic missing to limit how much position interpolation is allowed before requiring another good detection lock.

It's possible to use pipewire to look at the sensor feed if you build with that available. OpenHMD will then produce a pipewire video source that shares a debug view.