thaytan/OpenHMD #18: Literature research

witty91, 2020-10-23:
Hi there,
first of all I really enjoy this coming along so nicely! Since I do not have a regular Rift but only a Rift S I have started looking into this code and simultaneously papers on tracking and specifically inside-out tracking. I'm not yet too deep into the literature, but I thought that it might be useful to collect links to interesting publications on this here (since I assume after the Rift the next logical step would be work on the Rift S). 
For example, this appears to give a nice overview but not a lot of technical details:
https://onlinelibrary.wiley.com/doi/abs/10.1002/j.2637-496X.2017.tb00962.x
So, would this be of interest/helpful?

thaytan, 2020-10-25:
There is someone that's been on IRC a few times, looking into adding inside-out tracking to the Rift S headset. They pop up every now and then, but don't have any code public yet afaik.

As for papers to read, here's a few:

https://www.academia.edu/23959696/Visual_Odometry_Tutorial_ (Visual Odometry [Tutorial])

https://arxiv.org/pdf/1610.05949.pdf (Visual-Inertial  Monocular  SLAM  with  Map  Reuse)
https://intra.ece.ucr.edu/~mourikis/papers/Li2012-ICRA.pdf (Improving the Accuracy of EKF-Based Visual-Inertial Odometry)
https://www.jstage.jst.go.jp/article/transinf/E93.D/1/E93.D_1_152/_article (Real-Time Estimation of Fast Egomotion with Feature Classification Using Compound Omnidirectional Vision Sensor)
https://www.researchgate.net/publication/262378341_2-Point-based_Outlier_Rejection_for_Camera-IMU_Systems_with_applications_to_Micro_Aerial_Vehicles (2-Point-based Outlier Rejection for Camera-IMU Systems with applications to Micro Aerial Vehicles)

witty91, 2020-10-26:
Well, thank you very much! Seems like I'll have some reading to do :-D
Here's one more thing: 
[a bachelor's thesis on outside-in-tracking with selfbuilt hardware](http://resolver.tudelft.nl/uuid:28188648-0324-436d-8b63-700442963d24)

closed 2021-01-24 by thaytan