thaytan/OpenHMD #19: Unable to determine if this fork is any better for the DK2 than master.

LobotomyTH, 2020-11-22:
Hey, I wasn't sure where else to put this, but I'm interested in keeping the Oculus Development Kits alive, for the sake of archival, as well as the secondary bonus of one less piece of electronics going into a landfill. I have several of these HMDs, and have been thinking of giving a few to friends and OpenHMD contributors.

With that out of the way, I have been following your work on YouTube with the DK2/CV1, and I've been wondering if this is the right fork to use if I want at least _some_ rudimentary positional tracking? I realize that all of the Constellation reverse engineering is relatively new, and am aware of the unfinished elements in the tracking loop.

Additionally, I have seen other implementations of this from a very long time ago, but if I'm barking up the right tree with DK2 positional tracking, what is the possibility of using multiple cameras, for 360 degree tracking, or a wider tracking field? I'm sure that with the right jack splitter, it would be technically possible.

thaytan, 2020-11-22:
git master doesn't have any positional tracking support - only 3DOF, so if you want to try out the positional tracking this is the place for now.

There's nothing in the code that assumes there's only 1 camera for the DK2. I suspect it would work OK if you split the sync signal to multiple cameras. There's some work to do around the camera setup calibration - at the moment it calculates a calibration each time OpenHMD runs, and sometimes gets it wrong leading to spurious tracking errors.

The development branch to try is `rift-correspondence-search`

LobotomyTH, 2020-11-22:
Awesome, great to know. One last question, and maybe this would be one for the git master, but do either versions support the magnetometer in the DK2? It really doesn't matter a whole lot, because Oculus managed to get on just fine without it in the CV1, but it's just a fleeting curiosity of mine.

thaytan, 2020-11-23:
The DK2 mag is read, but the OpenHMD fusion code ignores magnetometer inputs. I'd guess one of the reasons Oculus dropped it in the CV1 is that mag input is hard to use indoors - there's lots of metal around that makes the magnetic field highly variable and affects the magnetometer readings accordingly.

thaytan, 2021-01-24:
Joey made a demo video of the https://github.com/thaytan/OpenHMD/tree/rift-kalman-filter branch doing positional tracking on a DK2 at https://www.youtube.com/watch?v=3AdmS3vy7ZE 

I think we'll start looking at merging some of this work back to upstream OpenHMD soon

closed 2021-01-24 by thaytan