thaytan/OpenHMD #21: Can't access CV1 cameras without sudo

hanaral, 2020-12-04:
Which means no steamVR testing without workarounds... Are there any updated permissions I need to setup to let my user and Steam access the cameras?

thaytan, 2020-12-05:
I'm using the xrhardware rules on my system, which include:

```
ATTRS{idVendor}=="2833", ATTRS{idProduct}=="0211", TAG+="uaccess", ENV{ID_xrhardware}="1"
```

You'll need some variant of that in an `/etc/udev/rules.d/` file

hanaral, 2020-12-05:
@thaytan even with the xr-hardware rules I set up it seems to not want to do anything without sudo. I’ve added the extra bit in your rules to no avail, I’ll try installing the xr-hardware from git

thaytan, 2020-12-05:
Can you post the output of your SteamVR-OpenHMD configure / build?

hanaral, 2020-12-06:
> Can you post the output of your SteamVR-OpenHMD configure / build?

Here's the output from meson, it never seems to generate the required /build/bin folder

```
han@elementary-os:~/SteamVR-OpenHMD$ meson build
The Meson build system
Version: 0.56.0
Source dir: /home/han/SteamVR-OpenHMD
Build dir: /home/han/SteamVR-OpenHMD/build
Build type: native build
Project name: steamvr-openhmd
Project version: 0.0.1
C++ compiler for the host machine: c++ (gcc 9.3.0 "c++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0")
C++ linker for the host machine: c++ ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64

|Executing subproject openhmd method meson 
|
|Project name: openhmd
|Project version: 0.3.0
|C compiler for the host machine: cc (gcc 9.3.0 "cc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0")
|C linker for the host machine: cc ld.bfd 2.34
|C++ compiler for the host machine: c++ (gcc 9.3.0 "c++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0")
|C++ linker for the host machine: c++ ld.bfd 2.34
|Library m found: YES
|Found pkg-config: /usr/bin/pkg-config (0.29.1)
|Run-time dependency hidapi-libusb found: YES 0.9.0
|Run-time dependency threads found: YES
|Run-time dependency libusb-1.0 found: YES 1.0.23
|Run-time dependency opencv4 found: YES 4.2.0
|Build targets in project: 3
|Subproject openhmd finished.

Dependency threads found: YES unknown (cached)
Build targets in project: 4

steamvr-openhmd 0.0.1

  Subprojects
    openhmd: YES

Found ninja-1.10.0 at /usr/bin/ninja
han@elementary-os:~/SteamVR-OpenHMD$ cd build/ && ninja
[18/59] Compiling C object subprojects/openhmd/libopenhmd.a.p/src_drv_oculus_rift_rift-tracker.c.o
../subprojects/openhmd/src/drv_oculus_rift/rift-tracker.c: In function ‘rift_tracker_free’:
../subprojects/openhmd/src/drv_oculus_rift/rift-tracker.c:218:4: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  218 |    if (dev->debug_metadata != NULL)
      |    ^~
../subprojects/openhmd/src/drv_oculus_rift/rift-tracker.c:220:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  220 |   ohmd_destroy_mutex (dev->device_lock);
      |   ^~~~~~~~~~~~~~~~~~
[53/59] Compiling C++ object driver_openhmd.so.0.0.1.p/driverlog.cpp.o
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:2126:8: warning: ‘class vr::IVRSettings’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2126 |  class IVRSettings
      |        ^~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2590:7: warning: ‘class vr::ITrackedDeviceServerDriver’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2590 | class ITrackedDeviceServerDriver
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2640:8: warning: ‘class vr::IVRDisplayComponent’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2640 |  class IVRDisplayComponent
      |        ^~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2687:8: warning: ‘class vr::IVRDriverDirectModeComponent’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2687 |  class IVRDriverDirectModeComponent
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2751:8: warning: ‘class vr::ICameraVideoSinkCallback’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2751 |  class ICameraVideoSinkCallback
[... 217 lines cut ...]
[ 66%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_wmr/packet.c.o
[ 68%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_psvr/psvr.c.o
[ 70%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_psvr/packet.c.o
[ 72%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_htc_vive/vive.c.o
[ 74%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_htc_vive/packet.c.o
[ 76%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_nolo/nolo.c.o
[ 78%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_nolo/packet.c.o
[ 80%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_3glasses/xgvr.c.o
[ 82%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_3glasses/packet.c.o
[ 84%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_vrtek/vrtek.c.o
[ 86%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_vrtek/packet.c.o
[ 88%] Building C object subprojects/openhmd/CMakeFiles/openhmd.dir/src/drv_external/external.c.o
[ 90%] Linking C static library libopenhmd.a
[ 90%] Built target openhmd
Scanning dependencies of target driver_openhmd
[ 92%] Building CXX object CMakeFiles/driver_openhmd.dir/driverlog.cpp.o
[ 94%] Building CXX object CMakeFiles/driver_openhmd.dir/driver_openhmd.cpp.o
[ 96%] Linking CXX shared library bin/linux64/driver_openhmd.so
[ 96%] Built target driver_openhmd
Scanning dependencies of target simple
[ 98%] Building C object subprojects/openhmd/examples/simple/CMakeFiles/simple.dir/simple.c.o
[100%] Linking C executable simple
[100%] Built target simple
han@elementary-os:~/SteamVR-OpenHMD/build$ 
```

thaytan, 2020-12-06:
Ah right - with the meson build you need to run `install_files_to_build.sh` manually after compiling to copy the files into the bin dir.

It's possible to get meson to do that, but noone has spent the time to implement it yet.

hanaral, 2020-12-06:
Alright, but steamvr seems to not detect any headset still. Even with Cmake the camera light comes on, yet with meson both headset and camera look dead (the demos work fine)

thaytan, 2020-12-07:
Nice! Want to make an MR for that? :)

hanaral, 2020-12-07:
@thaytan [done](https://github.com/ChristophHaag/SteamVR-OpenHMD/pull/47)

closed 2020-12-07 by hanaral