thaytan/OpenHMD #22: Fix lens distortion in CV1

hanaral, 2020-12-13:
I can’t attest for the Rift S, but I can say for sure that the image looks warped towards the centre in steamVR. There is an open issue on the main OpenHMD repo for the same thing outside of steamVR, so I presume it’s not on the steamVR side.

Danny-Dunn, 2021-08-01:
I've been working on some hand-calibration of my own and so far have found a config that works fairly well in the centre but isn't great closer to the edges, which I personally find more bearable.

A,B,C,D: `0.076, 0.060, 0.000, 0.819`

mattipee, 2021-08-17:
I find these improve the center of the image in my CV1, too, particularly the changes to values B and C. Thanks.

(For reference, originals are "0.098, .324, -0.241, 0.819")

I had a quick play and perhaps leaving the A value unchanged or even increasing to 0.15 might yield less edge distortion, to my eye, at first glance.

I will almost certainly now run with these modified values locally, thanks again.

thaytan, 2021-09-01:
I had a go at calculating the correct distortion with some through-the-lens photos. I came up with these values, which seem pretty close for me: `0.269, -0.25, 0.178, 0.803`

thaytan, 2021-09-01:
I also did some chromatic aberration measurements and got:
```
ohmd_set_universal_distortion_k(&(hmd_dev->base.properties), 0.269, -0.25, 0.178, 0.803);
ohmd_set_universal_aberration_k(&(hmd_dev->base.properties), 0.9992107, 1.0, 1.0120361);
```

These are now the values used on the `rift-kalman-filter` branch.

Danny-Dunn, 2021-11-06:
I found these parameters were better than the previous, but still had issues towards the center, where barrel distortion appears

thaytan, 2021-11-07:
It's not just you - after some time with the new calibration I agree it's not quite right yet.

It would make sense for there to be calibration info in the headset (to give exact screen positioning relative to the lenses), but if so I don't know where it's stored unfortunately.

Danny-Dunn, 2021-11-07:
> It's not just you - after some time with the new calibration I agree it's not quite right yet.
> 
> It would make sense for there to be calibration info in the headset (to give exact screen positioning relative to the lenses), but if so I don't know where it's stored unfortunately.

Is some help needed with reverse-engineering/USB packet capture of the official oculus software to figure this out(as well as prox sensor reports #36)?

thaytan, 2021-11-07:
If you're up for it, go for it. If you can find them, that'd help a lot.

Danny-Dunn, 2022-01-12:
Is there a good place to start with known information about the USB/HID protocol?

thaytan, 2022-01-12:
The only documentation is the driver code and/or jump on IRC and ask pH5 and me.