thaytan/OpenHMD #23: Cannot use features from non-enabled language CXX

NBorget, 2020-12-20:
Hi,
This is not realy an issue but I wanted to share this info. I don't know if there is any other place to share.
I was facing an issue with setting up this project using cmake.

`[main] Building folder: OpenHMD 
[main] Configuring folder: OpenHMD 
[cmake] Configuring done
[cmake] CMake Error in CMakeLists.txt:
[cmake]   Cannot use features from non-enabled language CXX
[cmake] 
[cmake] 
[cms-driver] Error during CMake configure: 
	Error: Failed to compute build system.
`
I solved this by adding CXX to the project in CMakeLists.txt
`project(openhmd C CXX)`

Now I'm able to setup and debug this project in VSCode.
In VSCode I use these extensions:
ms-vscode.cpptools
ms-vscode.cmake-tools

thaytan, 2020-12-20:
That's a new error - perhaps it only shows up with the VScode backend. Fixing shortly - thanks for the report.

If it happens with the upstream OpenHMD build too, you might want to report it there

NBorget, 2020-12-20:
Thanks for the update!

closed 2020-12-20 by NBorget