thaytan/OpenHMD #24: No positional information when building using CMake

NBorget, 2020-12-20:
When I build and make this project using CMake I don't receive positional information from my Rift CV1.
openhmd_simple_example output:
`rotation quat:           0.064488 0.278859 -0.003097 0.958160 
position vec:            0.000000 0.000000 0.000000 
controls state:          0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000`

When I use the Meson/Ninja flow, I do get this information:
`rotation quat:           0.086947 0.005303 -0.039788 0.995404 
position vec:            -1.170820 0.263225 -0.440830 
controls state:          0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000`

System:
Debian 5.8.10-1~bpo10+1 (2020-09-26) x86_64 GNU/Linux
Oculus Rift CV1

Am i missing steps in the CMake process?

thaytan, 2020-12-20:
`Am i missing steps in the CMake process?`

Pretty hard to tell from the provided information :)

I don't test the CMake build. I fixed it up so it at least compiles the other day, but didn't actually test it works.

NBorget, 2020-12-20:
Hich info can I provide?
I'm using the [rift-correspondence-search] branch. Only building with this flags :
OPENHMD_DRIVER_OCULUS_RIFT=ON
OPENHMD_EXAMPLE_SD=ON
OPENHMD_EXAMPLE_SIMPLE=ON

[CMakeCache.txt](https://github.com/thaytan/OpenHMD/files/5721068/CMakeCache.txt)

NBorget, 2020-12-20:
Thanks!
This fixed the issue.

closed 2020-12-20 by NBorget