thaytan/OpenHMD #32: Remove OpenCV dependency

thaytan, 2021-03-17:
A few OpenCV-supplied functions are critical to the positional tracking, but not many. The main one is RANSAC PnP + Levenberg-Marquardt pose refinement. That could be rewritten based on the integrated Lambatwist P3P implementation.

The others that need replacing are for projecting 3D LED positions to 2D, and undistorting/correcting blobs in captured frames to compensate for lens distortion. Both should be fairly easy to extract from OpenCV or reimplement.

gadLinux, 2023-10-15:
Why? Just in case you can do it better has some sense. But they are implemented and correctly. Let's use them

thaytan, 2023-10-16:
just because it's a big dependency and the tracking only needs a couple of small functions from it