thaytan/OpenHMD #33: Rift/Nvidia HMD standby mode when a game is launched

midget3111, 2021-03-22:
SteamVR home works great, but when ever I launch a game I'll see it for a couple of seconds in my Rift HMD, and then the headset goes into standby mode. Looking through the SteamVR logs I get the following -  

`"System] Unknown Transition from 'SteamVRSystemState_Ready' to 'SteamVRSystemState_Standby'."`

Using the latest rift-Kalman-filter branch in release mode on Arch with a GTX 1070. Headset and sensors are all connected via USB 3.0. 

This isn't urgent, just curious. I'll be updating this issue with relevant info and if I find a fix!

thaytan, 2021-03-22:
The only thing I can think of is that if the OpenHMD thread doesn't get serviced for too long, it will miss sending the watchdog keepalive message to the headset and the headset will switch off. That timeout is supposed to be 10 seconds though.

midget3111, 2021-03-22:
I'll try changing the nice value and reporting back!