thaytan/OpenHMD #38: Compiling with CMake doesn't work on Manjaro

Gladox114, 2021-05-29:
after the "make" command it can't find some libraries and can't compile it.
https://hastebin.com/zuturokiza.sql

Compiling meson with SteamVR-OpenHMD somehow doesn't work for SteamVR that's why I wanted to try it with CMake

143mailliw, 2021-05-30:
What's the problem you're having with meson?

thaytan, 2021-05-31:
I'm not sure what's going on there. It seems likely there's a libvtk devel package missing from your system, which would be a Manjaro packaging bug since their opencv build depends on it.

Gladox114, 2021-06-01:
> What's the problem you're having with meson?

I mean I can compile it without errors https://hastebin.com/abazulupig.cpp
but for some reason Steam can't find my device. 

Installing the AUR package "steamvr-openhmd-git" worked but it uses old openhmd drivers and not yours. I looked at the install script and it's using cmake so I thought why not trying it out with cmake...

Gladox114, 2021-06-01:
> I'm not sure what's going on there. It seems likely there's a libvtk devel package missing from your system, which would be a Manjaro packaging bug since their opencv build depends on it.

Because you wrote this I tried to search for "libvtk" in the software GUI and with the term "vtk" I found a package. Installed it and it worked without problems to compile with CMake. Not sure why it wasn't there in the first place because opencv was already installed (does opencv just use binaries when distributing? maybe this was the reason? I'm just a noob)

Gladox114, 2021-06-01:
So the compilation worked I should close this thread.

But I still didn't manage to make it work. But at the end I redid everything with cmake but copied commands from the AUR package that "SteamVR-Openhmd-git" used (the "install -Dm644..." commands) and then it finally worked. So the Meson build should work too.

closed 2021-06-01 by Gladox114