thaytan/OpenHMD #42: No display/tracking without sudo

zeskeertwee, 2021-09-04:
When running the openGL example, i get no tracking and the display stays off (proximity sensor is covered). However, when running the openGL example with sudo, the headset display turns on after both monitors going black for a second which is followed by the rift displaying the dektop and the monitor displaying the rift's view, tracking does work. I have installed the xr-hardware udev rules. I also have extended mode enabled as described [here](https://github.com/OpenHMD/OpenHMD/wiki/Xorg) (i'm using a nvidia GPU). Disabling extended more results in the rift's screen not turning on, and running without sudo provides no tracking, with sudo tracking works but i get the rift's view in a window (no going black).

Danny-Dunn, 2021-09-04:
My guess is that your distro-provided xr-hardware package is out of date.

Try installing this manually instead: [freedesktop.org GitLab xr-hardware/70-xrhardware.rules](https://gitlab.freedesktop.org/monado/utilities/xr-hardware/-/blob/main/70-xrhardware.rules)

zeskeertwee, 2021-09-05:
I have installed the xr-hardware from the repo, with `make && sudo make install`. I tested again and it did nothing without sudo (no tracking/display).

thaytan, 2021-09-05:
Did you reboot after installing? udev won't pick up new rules without either a reboot or a:
```
udevadm control --reload-rules
udevadm trigger
```

Take a look at https://github.com/OpenHMD/OpenHMD/wiki/Udev-rules-list for some other explanations that might help.

zeskeertwee, 2021-09-05:
after rebooting, i have the same results as before, except without sudo needed (extended mode shows desktop in headset, headset view in window / direct mode headset display stays off and headset view in window).

thaytan, 2021-09-06:
The OpenHMD OpenGL example doesn't implement direct mode support, so only works in Extended mode - you need to start it, then drag the window to the headset screen and fullscreen it by pressing F1.

If, on the other hand you want to use your headset with SteamVR or monado, you want it in Direct mode and to build the SteamVR-OpenHMD driver.

OOC, which headset are you testing?

zeskeertwee, 2021-09-06:
Oh thanks, Il try building steamvr-openhmd and testing steamvr. l'm using an Oculus rift CV1

thaytan, 2021-09-06:
https://noraisin.net/diary/?page_id=1048 is the HOWTO I wrote for testing SteamVR with my tracking branches

zeskeertwee, 2021-09-10:
SteamVR works in direct mode!

closed 2021-09-10 by zeskeertwee