thaytan/OpenHMD #43: Display all warpy and weird (weird distortion correction)

Joemongus6969, 2021-10-16:
My headset display is a little weird looking, kinda like how the display mirroring of old oculus headsets would look on a video or something, but in the headset. 

after playing with the distortion correction settings in rift.c (in src/drv_oculus_rift) around line 1233 it looks a little better, to the point where it's usable, but it's not perfect. I have no idea how to measure this, so I can't really make it perfect (not to mention how I barely understand what is going on there). 

if anyone else is having this problem, I'd recommend you try this, it will probably give you better results than you had before. if anyone knows how to measure and better configure this stuff, please type it in the comments.

thaytan, 2021-10-17:
The distortion parameters that are in there are ones I measured on my headset fairly recently (https://github.com/thaytan/OpenHMD/issues/22). I thought my measurements were better for everyone, but it seems not.

Which values have you found work better for you?

Joemongus6969, 2021-10-17:
I'm still playing with it, I can only do it so often because I find it very nauseating. I've been trying to figure out what all the values do, for instance; I've found the first value changes the distance of the virtual screen, but I'm not sure what the other three do.

thaytan, 2021-10-17:
There's nothing so straightforward about the parameters. They are coefficients of a polynomial that defines a distortion curve - https://wiki.panotools.org/Lens_correction_model