thaytan/OpenHMD #44: Stuttering, loss of tracking when haptics are activated

Danny-Dunn, 2021-11-06:
On recent versions(past month or 2, cant remember exactly when) the system began stuttering whenever haptics were activated, this is particularly obvious in Beat Saber, where both controllers will lose tracking within 20-30 seconds of starting the level. The entire system(OpenHMD, SteamVR, the game) appears to momentarily slow down or lock up whenever haptics are triggered. Games play fine with haptics disabled.

thaytan, 2021-11-07:
I haven't had anything like that.

I did find that constant rumbling could confuse the accelerometer/pose matching and lead to tracking losses, but I made changes a week or two ago that fixed that on my machine. I can stand with sabers crossed so the controllers rumble constantly without losing tracking now.

Are you compiling in release mode? What's your system spec?

Danny-Dunn, 2021-11-07:
Tried compiling fresh using [these](https://noraisin.net/diary/?page_id=1048) instructions.

My specs:
AMD Ryzen 7 2700 (overclocked to 3.9GHz) CPU
16GB 3200MHz Corsair Vengeance RGB Pro DDR4 Memory
Asus ROG Strix B450-F Gaming II Motherboard
Asus ROG Strix AMD RX570(overclocked) Graphics Card
About 1.2TB of SSD storage

Danny-Dunn, 2021-11-07:
> I haven't had anything like that.
>
> I did find that constant rumbling could confuse the accelerometer/pose matching and lead to tracking losses, but I made changes a week or two ago that fixed that [...]

Interesting, I never had issues with the rumbling causing tracking losses even before those changes.

Danny-Dunn, 2021-11-07:
SteamVR shows this output 
`Sun Nov 07 2021 09:51:48.523669 - Invalid input type click for controller (/user/head/proximity)
Sun Nov 07 2021 09:51:48.557234 - Invalid input type touch for controller (/user/hand/left/input/joystick)
Sun Nov 07 2021 09:51:48.557474 - Invalid input type touch for controller (/user/hand/left/input/x)
Sun Nov 07 2021 09:51:48.557571 - Invalid input type touch for controller (/user/hand/left/input/y)
Sun Nov 07 2021 09:51:48.557746 - Invalid input type touch for controller (/user/hand/right/input/joystick)
Sun Nov 07 2021 09:51:48.557954 - Invalid input type touch for controller (/user/hand/right/input/a)
Sun Nov 07 2021 09:51:48.558093 - Invalid input type touch for controller (/user/hand/right/input/b)
Sun Nov 07 2021 09:51:48.558724 - Invalid input type click for controller (/user/head/proximity)`

thaytan, 2021-11-07:
Those machine specs seem fine, and the messages from SteamVR are noisy but harmless I think (I get them too, they're something to do with the input mapping files).

My only guess is that the problem is at the USB traffic level, since the only real cost of starting and stopping haptics is sending a radio command. You could try rearranging which USB ports you're plugged into, otherwise you might need to do some git bisecting back and see if you can identify when it started.

Danny-Dunn, 2021-11-17:
The haptics problem no longer exists, even in versions I though it did. Strange.

thaytan, 2022-01-19:
Closing. Please reopen if you see problems again.

closed 2022-01-19 by thaytan