thaytan/OpenHMD #45: No tracking in Zenith

TheGeka, 2022-06-15:
First of all thank you for your work the driver works amazing except a slight delay between head movement and the movement visible inside the headset.
Now to the actual issue, Tracking works great everywhere except in Zenith or DeoVr. As soon as the app is done starting the controllers stop tracking, if I open the steamvr overlay in-game they continue to work just fine.

OS: Kubuntu 21.10
Kernel version: 5.17.0-8.1-liquorix-amd64
HMD: Oculus CV1
OpenHMD: 489481c38387066a30dd5b5a7a48a76ecf129198
SteamVr: 1.22.13

Danny-Dunn, 2022-06-15:
You may need to set the steamvr controller config to pretend that it is a different kind of controller.

TheGeka, 2022-06-19:
I tried using the vive binding  but that also did not help still the same issue