thaytan/OpenHMD #48: Does the actual HMD have positional tracking?

NabsiYa, 2022-09-22:
I was wondering if the HMD does have Positional tracking because it's definitely not enabled by default I tried using OpenHMD in SteamVR but it was barely bearable because the HMD is not being tracked with the Sensors it just felt that there is huge latency and it most of the time always was kinda tilted so I was not looking straight is there fix for this? This issue probably would be better to put into thaytan/SteamVR-OpenHMD but I am pretty sure this is not issue within the driver itself.

thaytan, 2022-09-24:
Take a look at https://noraisin.net/diary/?page_id=1048

NabsiYa, 2022-12-22:
back to this, I've verified that the HMD infact does not have Positional tracking...

device 0
  vendor:  Oculus VR, Inc.
  product: Rift (CV1)
  path:    5-1.1:1.0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

Danny-Dunn, 2022-12-22:
Have you followed the steps that @thaytan outlines in the blog post linked above? Positional tracking is experimental and requires compiling yourself.

NabsiYa, 2022-12-22:
Yeah I followed the blog post above even using the same operating system as the blog post describes yet it does not work.