thaytan/OpenHMD #49: Unable to turn on the display on the Oculus CV1

LameMonster82, 2022-10-24:
Hi, for some time I have been trying to get my Oculus CV1 to display ANYTHING on its display.
I have followed the guide from the diary https://noraisin.net/diary/?page_id=1048
And I have installed all equivalent packages for my Arch distro including xr-hardware 1.1.0.

However whenever I run SteamVR or monado-service they seem to start fine including CV1 tracking but the display on the headset remains disabled and the led stays orange.
`xrandr` does not detect the headset being plugged it at all.
Tracking seems to function as expected.

If I add `AllowHMD` to my Xorg config that allows me to see the headset being plugged in but only when OpenHMD is running. When it is running the display sometimes appears as "disabled" in the NVIDIA Settings and xrandr. Enabling it and changing any possible settings does not affect the HMD at all. It still displays nothing with orange led. However the PC does seem to detect the display and act like it works as I'm able to drag windows to it (although I have no idea how it looks like)

I have tried using oculus-udev package for the DK1/2 and writing my own udev rules but nothing has worked so far.
I am sure that the HDMI port and the HMD works in windows.

I am using the NVIDIA GTX 1050 Gpu with the 520.56.06 driver version on a pure Arch system.

thaytan, 2022-10-25:
If the tracking works, then the USB devices are accessible - so your udev rules are OK.

Other people have reported problems with their CV1 display not lighting up and that it worked after downgrading their kernel and/or nvidia driver. Like here: https://github.com/OpenHMD/OpenHMD/issues/340#issuecomment-1193042375

LameMonster82, 2022-10-25:
Can confirm that the 510 nvidia driver and the kernel 5.15 seem to work. The issue may be from upstream as I doubt that there is much of a change between 510 and 520 driver version. It will be a good idea to find the root cause of it.

closed 2022-10-25 by LameMonster82