thaytan/OpenHMD #52: More of a question then anything.

jetblade, 2024-05-13:
With this software when using the oculus rift will I be-able to use more then 1 VRObject?

thaytan, 2024-05-14:
I don't know what that means

jetblade, 2024-05-14:
> I don't know what that means

I was told that your software might support more then 1 vrobject for the oculus rift, for the oculus rift it supports connecting an additional controller and are able to use it as a ingame object like a gun, model car for example but also allows third party programs to change its usage, in my instance I use Oculus Touch Steam Link, which allows me to use the third controller as a tracker, but do to the oculus software they only set it up to allow 1 vroject, even though the hardware can support more.

thaytan, 2024-05-15:
Ah, you mean pairing more controllers. I haven't tried.

I think the answer is likely no: OpenHMD doesn't have any code for pairing controllers to the headset - so that already has to be done in Windows where they'll only let you pair 3 things. It might be a fundamental limitation of the headset firmware or the radio protocol.

jetblade, 2024-05-16:
> Ah, you mean pairing more controllers. I haven't tried.
>
> I think the answer is likely no: OpenHMD doesn't have any code for pairing controllers to the headset - so that already has to be done in [...]

I see, thank you

jetblade, 2024-05-17:
> Ah, you mean pairing more controllers. I haven't tried.
>
> I think the answer is likely no: OpenHMD doesn't have any code for pairing controllers to the headset - so that already has to be done in [...]

Question, what do you mean by (be done in Windows) the pairing system is done in the oculus program, or am I misunderstanding?

thaytan, 2024-05-19:
> > Ah, you mean pairing more controllers. I haven't tried.
> > I think the answer is likely no: OpenHMD doesn't have any code for pairing controllers to the headset - so that already has to be done in Windows where they'll only let you pair 3 things. It might be a fundamental limitation of the headset firmware or the radio protocol.
> 
> Question, what do you mean by (be done in Windows) the pairing system is done in the oculus software, or am I misunderstanding?

The only software that can pair controllers to the headset is the Oculus software, running on Windows - and that only lets you pair 3 devices, so even if the hardware supports more internally (unknown), 3 is the achievable limit.

jetblade, 2024-05-20:
> The only software that can pair controllers to the headset is the Oculus software, running on Windows - and that only lets you pair 3 devices, so even if the hardware supports more internally [...]

I get it now thank you, and I'm guessing its going to be extremely hard to modify the Oculus software? how much do you think it would cost?

thaytan, 2024-05-22:
> I get it now thank you, and I'm guessing its going to be extremely hard to modify the Oculus software? how much do you think it would cost?

I wouldn't even try. If I wanted to figure out how many devices the headset can handle, I'd start by deciphering the USB pairing protocol and writing a utility. I suspect there's a reason that they chose 3 as a limit though. It might be all the devices the headset to controller radio protocol can manage

jetblade, 2024-05-22:
> I wouldn't even try. If I wanted to figure out how many devices the headset can handle, I'd start by deciphering the USB pairing protocol and writing a utility. I suspect there's a reason that they [...]

F, is there a way to use 2 of the same headset so I can bypass this problem?