thaytan/OpenHMD #53: Future Directions: Monado

Doomsdayrs, 2024-08-24:
OpenHMD has long since become unmaintained.

The future for the original Rift CV1, for those who do still have one, or those who wish to keep the old hardware alive and usable, is in my opinion, to port the code to Monado.

This would be a sizeable investment, that would take *months* of time for even a build to work.

Though, If you wish to be extra, Do it all in Rust for memory safety.

---

@thaytan Would you like to pin this issue, to direct any prospective developers into the future.

thaytan, 2024-08-30:
Yes, porting to Monado is the next step. I don't think it would take months. Rewriting in Rust would require first adding Rust build support and bindings glue into Monado - even before getting to the driver itself. It's likely to make things easier in the long run, but I'd probably just port the existing code first.

Doomsdayrs, 2024-08-30:
> Yes, porting to Monado is the next step. I don't think it would take months. 

I said months, thinking of someone unfamiliar with the code.

> Rewriting in Rust would require first adding Rust build support and bindings glue into Monado - even before getting to the driver itself.

I would be lying if I said I didn't mention it because I was interested in figuring that out.

Mhowser, 2024-10-16:
is there a repo we can track the progress on the rewrite?

thaytan, 2024-10-19:
There's nothing yet. I have made a start on porting the driver but it's a body of work. I'll push it to https://gitlab.freedesktop.org/thaytan/monado/-/tree/oculus-rift-cv1 once I have something functional